PR2/pr2_simulator #138: Linking with Gazebo in pr2_gazebo causes a build error in another package

pazeshun, 2018-03-13:
Using `find_package` to gazebo in https://github.com/PR2/pr2_simulator/blob/kinetic-devel/pr2_gazebo/CMakeLists.txt#L4 makes the following build error in `pr2eus`, which uses `find_package` to `pr2_gazebo` in https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/CMakeLists.txt#L50
```
CMake Error at /opt/ros/kinetic/share/pr2_gazebo/cmake/pr2_gazeboConfig.cmake:148 (message):
  Project 'pr2eus' tried to find library '-lpthread'.  The library is neither
  a target nor built/installed properly.  Did you compile project
  'pr2_gazebo'? Did you find_package() it before the subdirectory containing
  its code is included?
Call Stack (most recent call first):
  CMakeLists.txt:50 (find_package)
```
The similar problem is also reported in https://github.com/ros-planning/navigation/issues/612 and https://github.com/ros/catkin/issues/856

pazeshun, 2018-03-13:
I also found `pr2_gazebo_plugins` uses `find_package` to gazebo, which may cause other errors in other packages depending on `pr2_gazebo_plugins`.

pazeshun, 2018-03-13:
@k-okada 
Can I remove `find_package` to gazebo?
Or should I try your 'dirty fix' in https://github.com/ros-planning/navigation/issues/612#issuecomment-321715401

k-okada, 2018-03-13:
> Or should I try your 'dirty fix' in ros-planning/navigation#612 (comment)

prease provide more information, what package did you build, and what was
the error message.

pazeshun, 2018-03-13:
I built master branch of `pr2eus` on kinetic and got a build error as follows:
```
$ source /opt/ros/kinetic/setup.bash
$ mkdir -p test_ws/src && cd test_ws/src
$ git clone https://github.com/jsk-ros-pkg/jsk_pr2eus.git
$ cd ..
$ catkin build
-----------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/pazeshun/test_ws
-----------------------------------------------------------
Source Space:       [exists] /home/pazeshun/test_ws/src
Log Space:         [missing] /home/pazeshun/test_ws/logs
Build Space:        [exists] /home/pazeshun/test_ws/build
Devel Space:        [exists] /home/pazeshun/test_ws/devel
Install Space:      [unused] /home/pazeshun/test_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-----------------------------------------------------------
[build] Found '4' packages in 0.0 seconds.                                                          
[build] Updating package table.                                                                     
Starting  >>> catkin_tools_prebuild                                                                 
Finished  <<< catkin_tools_prebuild                [ 1.4 seconds ]                                  
Starting  >>> pr2eus                                                                                
____________________________________________________________________________________________________
Errors     << pr2eus:cmake /home/pazeshun/test_ws/logs/pr2eus/build.cmake.000.log                   
-- set EUSDIR to /opt/ros/kinetic/share/euslisp/jskeus/eus
-- set ARCHDIR to Linux64
CMake Error at /opt/ros/kinetic/share/pr2_gazebo/cmake/pr2_gazeboConfig.cmake:148 (message):
  Project 'pr2eus' tried to find library '-lpthread'.  The library is neither
  a target nor built/installed properly.  Did you compile project
  'pr2_gazebo'? Did you find_package() it before the subdirectory containing
  its code is included?
Call Stack (most recent call first):
  CMakeLists.txt:50 (find_package)

cd /home/pazeshun/test_ws/build/pr2eus; catkin build --get-env pr2eus | catkin env -si  /usr/bin/cmake /home/pazeshun/test_ws/src/jsk_pr2eus/pr2eus --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/pazeshun/test_ws/devel/.private/pr2eus -DCMAKE_INSTALL_PREFIX=/home/pazeshun/test_ws/install; cd -
....................................................................................................
Failed     << pr2eus:cmake                         [ Exited with code 1 ]                           
Failed    <<< pr2eus                               [ 1.7 seconds ]                                  
Abandoned <<< pr2eus_moveit                        [ Unrelated job failed ]                         
Abandoned <<< pr2eus_tutorials                     [ Unrelated job failed ]                         
[build] Summary: 1 of 4 packages succeeded.                                                         
[build]   Ignored:   1 packages were skipped or are blacklisted.                                    
[build]   Warnings:  None.                                                                          
[build]   Abandoned: 2 packages were abandoned.                                                     
[build]   Failed:    1 packages failed.                                                             
[build] Runtime: 3.1 seconds total.                                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

pazeshun, 2018-03-13:
In kinetic, the version of released `pr2_gazebo` is 2.0.9 and it includes `find_package` to gazebo added in 9125eed8848c.
Also, master branch of `pr2eus` includes `find_package` to `pr2_gazebo` added in https://github.com/jsk-ros-pkg/jsk_pr2eus/commit/dda04e21080e2f498e71c94f41f1eadabf427d92.
The reason why `pr2eus` released in kinetic is built successfully is that its version is 0.3.10 and it doesn't include `find_package` to `pr2_gazebo`.

pazeshun, 2018-03-13:
On the other hand, in indigo, released `pr2_gazebo` is 2.0.7 and doesn't include `find_package` to gazebo.
This causes master branch of `pr2eus` can be successfully built in indigo although it includes `find_package` to `pr2_gazebo`.

pazeshun, 2018-03-13:
As mentioned in https://github.com/ros/catkin/issues/856, `-lpthread` is passed in `gazebo_LIBRARIES` like `SDL_LIBRARY` in https://github.com/ros-planning/navigation/issues/612#issuecomment-321715401, which causes an error in `find_package`.

k-okada, 2018-03-16:
thank you for explanation,  #139 will solve this problem

PR #139 (2018-03-18): remove find_package(gazebo), to solve -pthread linking errors
Closes #138
closed 2018-03-18 by k-okada